yaleman/pygoodwe #27: How to use it?

JB1985, 2020-06-12:
Hello, 

I have a Goodwe hybrid inverter and want to use your pygoodwe. But I dont know how to use it?!

Can you please write a little HowTo?

yaleman, 2020-06-12:
Hi there, there's some example code in [test.py](https://github.com/yaleman/pygoodwe/blob/master/test.py), or alternatively I use pygoodwe in my [goodwe2pvoutput](https://github.com/yaleman/goodwe2pvoutput/blob/master/goodwe2pvoutput/bin/goodwe2pvoutput) script which pumps data to [PVOutput.org](https://PVOutput.org)

JB1985, 2020-06-19:
Thanks for your Script!

Do you know, how to get the data from house load and sell?

[image]

The other for PV and Battery i have find.

JB1985, 2020-06-19:
The "sell" seems to be "pmeter" but I dont know how to get the data. I am trying with:

`print(json.dumps(gw.data.get('pmeter'), indent=2))`

But the output is always NULL.

JB1985, 2020-06-20:
House load is "getLoadFlow". For "sell" -> "pmeter" i have open a pull request.

For me it works fine now!

closed 2020-06-20 by JB1985