yaleman/pygoodwe #31: API Access

RetiefTech, 2020-07-01:
Hi, got this error since yesterday:

WARNING:root:HTTPSConnectionPool(host='globalapi.sems.com.cn', port=443): Max retries exceeded with url: /api/v1/PowerStation/GetMonitorDetailByPowerstationId (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)')))

Could change the global_url to : 'https://semsportal.com:443/api/'
or add an option to switch between the 2 or try the 1 url then the other?

JB1985, 2020-07-05:
Here is the PR -> https://github.com/yaleman/pygoodwe/pull/32

yaleman, 2020-07-06:
Thanks for raising this folks, I've fixed it here: 5f8cbe8a74064120f88098e34c76bf4c6c819eaf

Apologies for the delay!

closed 2020-07-06 by yaleman