yaleman/pygoodwe #38: Data issues with multiple inverters

wishie, 2020-08-29:
I will first explain my setup, and then the problems I am facing.

1) GoodWe GW5048-EM Hybrid Inverter with an eSTORE 10.3kWh LiFePO4 battery attached with 6.1kW of PV (Inverter 1)
2) GoodWe GW5048-EM Hybrid Inverter with with 6.6kW of PV and NO BATTERY attached. (Inverter 2)

To be clear, I think most of these may be issues with the portal, NOT your code, but I will mention it all here just in case.

When I was using just the first inverter, things were fine. Data being read was accurate. Adding the 2nd inverter to the mix made things a complete mess. Here are some of the issues:

1) SOC instantly halved (Inverter 1 SOC at 80%, Inverter 2 SOC at 0%) SEMS displays 40% SOC... even if its aware there is no battery on the 2nd inverter.
2) Power that I BUY from the grid is also doubled (I think the 'smart meters' are both measuring the same power coming into the property, and SEMS adds them together.. So if I am drawing 700W from the grid, its counted by both smart meters are shows 1400W)
3) With just Inverter 1 connected "SelfUseOfPV" used to include the power put into the battery.. this no longer seems to be the case, as sometimes I will have charged the battery from 10% to 100%, yet "SelfUseOfPV" is only 3kWh for the day (this is completely wrong.. it would be more like 15-20kWh)
4) Sometimes the portal shows the LOAD as something like 17W (as in the house is drawing 17w.. this is obviously incorrect)
5) Sometimes the portal shows I am EXPORTING/SELLING more than my PV is generating, even whilst still charging the battery, and feeding the LOAD of the house.. eg, PV is generating 10kW, battery is charging at 2.6kW, load is 1kW, exporting/selling 12kW..

Im sure there are more issues, but it seems that the SEMS Portal is NOT a reliable source at all currently. I will be discussing this with GoodWe tomorrow, but do not have high hopes of them quickly fixing these issues.

Do you know if there is a way to get data directly from the inverter itself? I used to use RS485->USB on my old ABB/PowerOne inverter.

yaleman, 2020-08-29:
Oh dear, that's really unfortunate on how they're presenting the data. I vaguely remember the data structures will show up in a "global" and a "per device" fashion (I've only got one inverter to test with) - maybe it's possible to dis-aggregate it that way? Try dumping the data out of `API.data.get('inverter')` (where API is the instance of the class) - there should be one per inverter.

As for RS-485, there is an interface and documentation available, you have to sign an NDA to get it though.

wishie, 2020-08-30:
Yes, it really is a mess.. I specifically got the 2nd GoodWe to make things easier on myself. I was wrong.

I currently just use json.dumps and read through all the data myself, trying to figure it out.. its just not pretty at all. It is a problem also with their portal. Take a look at this example from this morning.. https://imgur.com/gKKxuox

I have fired off an email to both GoodWe Australia (where I am from) and GoodWe China, in hopes they can make some sense of all of this.

wishie, 2020-08-30:
If you are interested to see how the data is shown for 2 inverters, I can send you a link to the output.

yaleman, 2020-08-30:
That'd be cool for my own testing - something like a `json.dumps(API.data)` would be great 👍🏻

wishie, 2020-08-30:
I will organise that shortly and send you an email if that's ok.

I see you are located in Brisbane.. I'm on Russell Island, just off Redland Bay.

wishie, 2020-08-30:
I sent an email earlier. Hopefully you got it and it didn't go to junk.

As I mentioned, the data is updated every minute by a cron job. I hope it's of use to you.

kempert, 2021-04-20:
Hello, i also have two Intvertes, one for the Batteries and One for the PV. Is her any way to load the right data? Is anywehre a list of the structure of the API? 

I Also have the Problem with the SOC wich ist "47%" and the Sells Power is the Output Power of the PV

yaleman, 2021-04-20:
Sorry @wishie I've still got your email in my backlog, but life got in the way.

@kempert it sounds like this is a GoodWe portal problem, but if you do a `API.getCurrentReadings()` then `self.data` (or the response from `API.getCurrentReadings()` will be the full API dataset straight from their servers (as a python dict).

yaleman, 2022-02-09:
As this doesn't seem to be an ongoing issue with the package, I'll close this out. Feel free to open a new ticket if there's any issues.

closed 2022-02-09 by yaleman