yaleman/pygoodwe #65: Errors by testing pygoodwe import in IoBroker

kempert, 2021-04-19:
I want import my Goodwe SEMS Data in IoBroker, so I used the work from https://forum.iobroker.net/topic/34626/goodwe-wechselrichter-und-iobroker. When im tested it following problem comes up. Im no specialist in python Syntax. 

```
pi@Raspberry-Pi4:~ $ python /home/pi/pygoodwe-master/test.py
Traceback (most recent call last):
  File "/home/pi/pygoodwe-master/test.py", line 5, in <module>
    from pygoodwe import SingleInverter, API
  File "/home/pi/pygoodwe-master/pygoodwe/__init__.py", line 19
    def __init__(self, system_id: str, account: str, password: str, **kwargs):
                                ^
SyntaxError: invalid syntax
```
I imported the master and moving the config.py.example to config.py and enter data.

Can anyone help me to use the Script?

yaleman, 2021-04-19:
What version of Python are you using? Typically that'll mean you're using a very old one.

James

kempert, 2021-04-20:
I have installaed follwing versions.

```
pi@Raspberry-Pi4:~ $ python --version
Python 2.7.16
pi@Raspberry-Pi4:~ $ python3 --version
Python 3.7.3
```

yaleman, 2021-04-20:
Do you get the same error when you run "python3 /home/pi/pygoodwe-master/test.py" ?

kempert, 2021-04-20:
oh yess that was the Problem with phyton3 it is load perfectly. THANKS

yaleman, 2021-04-20:
Glad to see it worked out - python 2 is most definitely legacy these days :)

closed 2021-04-20 by yaleman