yaleman/pygoodwe #91: Hangs when trying to retrieve daily data "getDayDetailedReadingsExcel()"

windcrusader, 2021-12-19:
I get an issue with the module hanging when trying to download the "spreadsheet" data. Is there another function that can download the full set of daily readings?

Below is the code snippet that hangs
`gw.getDayDetailedReadingsExcel()`

yaleman, 2021-12-20:
It looks like the call's failing for some reason, I'm having a look at it now.

yaleman, 2021-12-20:
Ok, digging into this - there was a call to `v1/ReportData/GetStationPowerDataFilePath` which would return the file download path... it's no longer doing this. Checking it in the web UI, they've moved to a new API endpoint which I can't infer from the rest of the API... which is a pretty painful problem 😢

yaleman, 2021-12-20:
... actually, they decided to return "Successful" instead of one of the previous options - "Success" and "success"... because consistency's for suckers. 😄

windcrusader, 2021-12-21:
Thanks for your efforts much appreciated. I did a bit of sleuthing on the api with a packet sniffer (postman). 
When I click the "prepare data" button on the web portal the following api gets posted:
https://semsportal.com/GopsApi/Post?s=HistoryData/QueryTargetByEquipmentType

Then on saving the excel file there are a series of calls to the api:
https://semsportal.com/GopsApi/Post?s=HistoryData/ExportExcelStationHistoryData
with payload: (I've xxx'd a couple of fields for privacy)
t {"api":"HistoryData/ExportExcelStationHistoryData","param":{"tm_content":{"qry_time_start":"2021-12-21 00:00","qry_time_end":"2021-12-21 19:48","times":5,"qry_status":0,"pws_historys":[{"id":"ce34a6a9-c0e3-44d1-bb68-xxxxxxxxx","pw_name":"xxxxxx","status":1,"pw_address":"xxxxxx","inverters":[{"sn":"95000EHR214W0097","name":"95000EHxxxxx","change_num":0,"change_type":0,"relation_sn":null,"relation_name":null,"status":1}]}],"targets":[{"target_key":"Vpv1"},{"target_key":"Vpv2"},{"target_key":"Ipv1"},{"target_key":"Ipv2"},{"target_key":"Vac1"},{"target_key":"Iac1"},{"target_key":"Fac1"},{"target_key":"Tempperature"},{"target_key":"HTotal"},{"target_key":"pv_etotal_l+pv_etotal_h*65536"},{"target_key":"Pmeter"},{"target_key":"Pbackup"},{"target_key":"Vload"},{"target_key":"Iload"},{"target_key":"VBattery1"},{"target_key":"IBattery1"},{"target_key":"Cbattery1"},{"target_key":"SOH"},{"target_key":"BMS_Temperature"},{"target_key":"BMS_Charge_I_Max"},{"target_key":"BMS_Discharge_I_Max"},{"target_key":"EDay_EXP_1"},{"target_key":"ETotal_EXP_1"}]}}} 

Then when clicking the download button I get:
https://semsportal.com/GopsApi/Post?s=HistoryData/GetStationHistoryDataFilePath
with payload:
str={"api":"HistoryData/GetStationHistoryDataFilePath","param":{"file_id":"8c2e576e-08a4-4983-b507-7xxxx"}} 

Hopefully that helps?

yaleman, 2021-12-21:
Ok, I found the page you're talking about - that one's very different to the one that's been used before in the `pygoodwe` code - either way I can't get past however they expect one to authenticate against that `GopsApi` endpoint.. it's clearly an ASP.net something, I'm passing the cookie from login and it's just not working.

```
{
'hasError': False,
'code': 100001,
'msg': 'No access, please log in.',
'data': None,
'components': {
    'para': None,
    'langVer': 125, 
    'timeSpan': 0,
    'api': 'http://localhost:82/api/v1/ReportData/GetPowerStationPowerReportByMonth', 
    'msgSocketAdr': ''}
}`
```

... if only they'd publish the details of their api 😶‍🌫️

PR #179 (2022-12-29): xls fixes thanks to @dredboi
fixes #91
closed 2022-12-29 by yaleman