yaleman/pygoodwe #95: Calls seem to fail when PV flow is 0

tywjohn, 2022-01-07:
I noticed at night when there is no generation, I don't even get data for other things like inverter temperature or voltage etc.
I have yet to investigate further. Unsure if this is the API returning no data when there is no generation or whether it is pygoodwe not handling it properly.

darylkell, 2022-01-07:
I don't know about you, but no generation = inverter is off

tywjohn, 2022-01-07:
That may be true. Solar is a whole new world to me so I didn't realise it turns itself off like that..

tywjohn, 2022-01-07:
A quick google search tells me you are right. Apologies.

closed 2022-01-07 by tywjohn

yaleman, 2022-01-07:
I've got a solar + battery inverter that always seems to send data, but I'm guessing solar-only ones won't.