yaleman/pygoodwe #148: data["v4"] = self.getLoadFlow()[0] results in Float object is not subscriptable error

Fuchio, 2022-10-05:
Hey, I noticed that the function GetDataPvOutput() results in float object is not subscriptable error because of the following lines:
  
data["v4"] = self.getLoadFlow()[0]  # power consumption  
data["v5"] = self.get_inverter_temperature()[0]  # inverter temperature  
data["v6"] = self.getVoltage()[0]  # voltage  

After removing the list indices of [0] it works fine given that getLoadFlow returns a single float rather than a list.

New code:

data["v4"] = self.getLoadFlow()  # power consumption  
data["v5"] = self.get_inverter_temperature()  # inverter temperature  
data["v6"] = self.getVoltage()  # voltage

yaleman, 2022-10-05:
That's ... curious, since [getLoadFlow()](https://github.com/yaleman/pygoodwe/blob/a63361cf514477b7b0441cb0b250823f2fea3145/pygoodwe/__init__.py#L417) should raise NotImplementedError 🤔

Fuchio, 2022-10-05:
To add some information; this is not in the source code of 0.0.16 release strange enough.

And @ your comment, the getLoadFlow does return floats

Fuchio, 2022-10-05:
Thanks! Just tested and works :). Also thanks for publishing the package.

closed 2022-10-05 by Fuchio

yaleman, 2022-10-05:
THanks for submitting the issue - [v0.0.1 is now live on pypi]( https://pypi.org/project/pygoodwe/0.1.0/)