yaleman/pygoodwe #158: Offline since 2022/10/21

ArthurLiao0816, 2022-10-25:
I have 4 sites. All of them showing offline at the same time while https://www.sems.com.cn/ showing working. Any answer or solution for this situation? 😭😭😭

ArthurLiao0816, 2022-10-25:
> Are you sure you're using the right API endpoint in the `pygoodwe` calls? That's the only thing I can think of - otherwise if you could share you code that'd be handy to try and get better help.

My script has been working for almost a year.  I pretty sure I'm using the right call.  Just wanna know if this kinda thing happend before.  I'm worried about semsportal stopping their service.

yaleman, 2022-10-26:
I haven't seen any issues but mine's not on the .cn portal. I'll close this as it's not an issue with the library, but folks are welcome to comment if they know more.

closed 2022-10-26 by yaleman

ArthurLiao0816, 2022-10-26:
Thanks for your info👍👍👍