yaleman/pygoodwe #175: It seems that the used URL no longer seems to be available from the command line

Henkeh, 2022-11-29:
When I try to readout some values using your library, I keep getting the following response:

ERROR:root:RequestException: 403 Client Error: Forbidden for url: https://semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId
ERROR:root:RequestException: 403 Client Error: Forbidden for url: https://semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId
ERROR:root:Failed to call GoodWe API url='https://semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId'

I have been using it successfully for quite a while now, but recently it stopped working. Do you have any idea how to solve this? Or am I doing something wrong?

yaleman, 2022-11-29:
How recently did it stop working? Today? A week ago? Either way, I'm really not sure. Some tests seem to be working from the repository test suite, but it's been a while since I've looked a this code..

klara31, 2022-11-29:
Last data I received was yesterday around 13.00 (CET)

klara31, 2022-11-29:
When globally changing the API endpoint URL in ```pygoodwe/__init__.py``` from v1 to v2, it works again

Henkeh, 2022-11-29:
That seems to work for me too.

yaleman, 2022-11-29:
That's not working for me, can you please submit a PR with your changes?

tywjohn, 2022-12-18:
I can confirm Gregor-at-GitHub's change works for me

yaleman, 2022-12-18:
Fixes have been merged, new release on [PyPI as 0.1.2](https://pypi.org/project/pygoodwe/0.1.2/) thanks for finding the update.

PR #177 (2022-12-18): Updating SEMS API Endpoints to V2
Fixes #175 thanks to @gregor-at-github for the fix.
closed 2022-12-18 by yaleman