yaleman/pygoodwe #178: Errors running example.py

dredboi, 2022-12-29:
I'm trying to run the example.py.. I've set up my details in config.py and I can see that is coming through.

Apologies if this is something obvious (I'm new to python), but it fails with this output:

`Single Inverter
ERROR:root:<RequestsCookieJar[]>
ERROR:root:Done login, token: {"uid": "dd8c67fc-5fc2-4080-a8bd-6bec7f38744d", "timestamp": 1672275548366, "token": "79bda1384c81bb66878184fc4132952e", "client": "ios", "version": "v2.0.4", "language": "en"}

dict_keys(['info', 'kpi', 'powercontrol_status', 'images', 'weather', 'inverter', 'hjgx', 'homKit', 'isTigo', 'tigoIntervalTimeMinute', 'smuggleInfo', 'hasPowerflow', 'powerflow', 'hasGridLoad', 'isParallelInventers', 'isEvCharge', 'evCharge', 'hasEnergeStatisticsCharts', 'energeStatisticsCharts', 'energeStatisticsTotals', 'soc', 'environmental', 'equipment'])
"0V/0A/0W"
Battery voltage is: 0.0
Getting XLS file
ERROR:root:payload: {'str': '{"api": "v2/PowerStation/ExportPowerstationPac", "param": {"date": "2022-12-28", "pw_id": "66b874d7-9758-48bb-baa2-742d60c301c4", "img_base64": "iVBORw0KGgoAAAANSUhEUgAAAAEAAAABCAYAAAAfFcSJAAAAAXNSR0IArs4c6QAAAA1JREFUGFdj+P///38ACfsD/QVDRcoAAAAASUVORK5CYII=", "img_width": 1, "img_height": 1, "is_removesoc": 0}}'}
ERROR:root:Failed to pull from https://au.semsportal.com/GopsApi/Post - response - {'language': 'zh_CN', 'function': None, 'hasError': True, 'msg': 'ver is not fund', 'code': 'UnknownVer', 'data': None, 'components': {'para': None, 'langVer': 176, 'timeSpan': 0, 'api': None, 'msgSocketAdr': None}}`

yaleman, 2022-12-29:
Oh, that's me being half done with trying to fix #91, there's just a bunch of things you can use in that file in general. I've removed that faulty code, thanks for the heads up.

dredboi, 2022-12-29:
Thanks for the quick response. I picked up the change but I still see this:

Single Inverter
ERROR:root:<RequestsCookieJar[]>
ERROR:root:Done login, token: {"uid": "dd8c67fc-5fc2-4080-a8bd-6bec7f38744d", "timestamp": 1672276753850, "token": "f094a951aaed085d5892c7f33d4b8aeb", "client": "ios", "version": "v2.0.4", "language": "en"}
Available fields in data: dict_keys(['info', 'kpi', 'powercontrol_status', 'images', 'weather', 'inverter', 'hjgx', 'homKit', 'isTigo', 'tigoIntervalTimeMinute', 'smuggleInfo', 'hasPowerflow', 'powerflow', 'hasGridLoad', 'isParallelInventers', 'isEvCharge', 'evCharge', 'hasEnergeStatisticsCharts', 'energeStatisticsCharts', 'energeStatisticsTotals', 'soc', 'environmental', 'equipment'])
"0V/0A/0W"
Battery voltage is: 0.0

yaleman, 2022-12-29:
Sorry for the confusion, but what are you expecting this library to do? It's a library designed to be used to get data out and use it in other applications.

dredboi, 2022-12-29:
No worries - I had a mistake my side. With your latest changes I can get info from the inverter now.

Thanks for the help - much appreciated. Now I just need to transfer my Java programming skills to python!

yaleman, 2022-12-29:
Anytime - if there's any issues with the library, feel free to reach out again! Welcome to Python. 😄

closed 2022-12-29 by yaleman

dredboi, 2022-12-29:
Not sure if this is useful for you: I saw you were trying to get the XLS file for a given date. 

I’ve managed this now with the code below:

date = "2022-09-05"
payload = {
    "date": date,
    "pw_id": inverter.system_id
}
data = inverter.call(
    "v1/PowerStation/ExportPowerstationPac", payload
)

payload = {
    "id": data
}
data = inverters.call(
    "v1/ReportData/GetStationPowerDataFilePath", payload
)

response = requests.get(data.get("file_path"))
open("solar_" + date + ".xls", "wb").write(response.content)

yaleman, 2022-12-29:
That's fantastic, thanks @dredboi - I've released 0.1.4 to include the code.