yaleman/pygoodwe #180: failed requests

Cravvler, 2022-12-30:
Hi there
wondering if anyone else experienced issues with grabbing data.
everything worked fine for me for arround 9 monts, but since start of december its not working anymore and i keep getting these errors:

ERROR:root:RequestException: 403 Client Error: Forbidden for url: https://semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId
ERROR:root:Failed to call GoodWe API url='https://semsportal.com/api/v1/PowerStation/GetMonitorDetailByPowerstationId'

seems to me like there has been some changes somewhere from SEMS side
anyone else experiencing this? 

BR, Cravvler

yaleman, 2022-12-30:
Upgrading to anything v0.1.2 and above should fix this.

Cravvler, 2022-12-31:
Hi again,
looks like a pull has solved the problem :)
maybe you can add the latest version to your release versions, since the older ones won't work anymore
thanks!

yaleman, 2023-01-01:
Oh, you're cloning it directly from GitHub? Why not use the PyPi package?

SebastianHanz, 2023-06-01:
Hello,
same problem here with pygoodwe 0.1.6.
Freshly installed (first time) yesterday, configured but not working. Exact the same output.
Any idea?
Thanks!

yaleman, 2023-06-02:
@SebastianHanz how'd you install the package, can you share the code you used to query the API? I can imagine there's a lot of ways this can go wrong.

SebastianHanz, 2023-07-26:
Sorry for late answer I was on holiday. I figured out my Problem... I grabbed the package from Github via `git clone https://github.com/yaleman/pygoodwe` and I received the whole thing out of the _master branch_. After playing around and changing some URLs in the init file from v1 to v2 it also worked for me, thats what also happened in the main branch i guess. So it was my own mistake, I used old version, now I got the newest version and everything runs great. Thank You!

yaleman, 2023-07-27:
No worries at all, I'll delete that branch since it shouldn't be around anymore - apologies for the confusion!

closed 2023-07-27 by yaleman