yaleman/pygoodwe #181: API logs me out on my iPhone

vozdeckyl, 2023-01-02:
When I use the API to fetch data from the SEMS portal, it logs me out from my iPhone app and I need to log back in. This happens whenever the data is fetched (so roughly every 30 seconds) so it is very annoying. Is there a way around this? Thanks!

yaleman, 2023-01-02:
That's ... strange, but that's definitely outside the scope of what this library could control - that'd be something on GoodWe's side.

WouterGritter, 2024-07-08:
I have the same issue, but it also happens when not using this API and logging in on two phones. Any new login seems to log out all existing sessions on their side.

yaleman, 2024-07-10:
Yep, it's their end not my library - closing this one 😄

closed 2024-07-10 by yaleman