yaleman/pygoodwe #182: Arguments not propagated in SingleInverter.get_current_readings

vozdeckyl, 2023-01-03:
The arguments passed to `SingleInverter.get_current_readings`:

https://github.com/yaleman/pygoodwe/blob/96f0872561e692137f8d3cf3f9e93865a5c5e6a9/pygoodwe/__init__.py#L438

are not forwarded to `API.get_current_readings`. When the number of `maxretries` is set to 1, it's still 5 due to the default value of 5 in `API.get_current_readings`. Also the recursive call should not happen in `SingleInverter.get_current_readings` as it already happens in `API.get_current_readings`.

Furthermore, the function should return data, but it returns `retval` set to `None`.

This is what the function should imho look like:

```
 def get_current_readings(
        self,
        raw: bool=True,
        retry: int=1,
        maxretries: int=5,
        delay: int=30,
        ) -> Any:
        """ grabs the data and makes sure self.data only has a single inverter """
        # update the data
        super().get_current_readings(raw, retry, maxretries, delay)
        # reduce self.data['inverter'] to a single dict from a list
        
        if self.data.get("inverter"):
            self.data["inverter"] = self.data["inverter"][0]
        return self.data
```

yaleman, 2023-01-03:
Thanks for picking this up, I've pushed v0.1.5 which includes the fixes.

vozdeckyl, 2023-01-03:
Hi @yaleman,
Thanks! Shouldn’t the if statement stay there? If the API call is not successful, an empty diction is returned and then it will throw an error.

yaleman, 2023-01-04:
the `if` is done in the call in `API` so that should be fine

yaleman, 2023-03-21:
@vozdeckyl is this still an issue?

commit 15ed3c42 (2023-01-04): Merge pull request #183 from yaleman/yaleman/issue182

fixes #182 - pass settings through, handle errors better
commit e5764747 (2023-01-04): fixes #182 - pass settings through, handle errors better
closed 2023-03-21 by vozdeckyl