yaleman/pygoodwe #198: test_pygoodwe.py

chrmuh, 2023-04-24:
Hi, thank you for your libary. I tried to install it, but when I run test_pygoodwe.py I only get the message:

test_pygoodwe.py::test_instantiate SKIPPED (Could not find data file 'testdata.json') [ 12%]
Skipped: Could not find data file 'testdata.json'

and so on for all the data. Can you may help me.
Have a nice Day!

yaleman, 2023-04-24:
Look in `tests/test_pygoodwe.py` - the `inverter` function is a fixture for all the tests and it loads config based on environment variables - so if you don't tell it what to do, it'll look for cached test data.

chrmuh, 2023-04-24:
Dank you for your fast response. I am new to json, so this is really hard to understand, but I also just found the example.py code, and I debugged it. So I saw, that your program is working fine, I received my data. But I cannot print it with print.
All your print lines are grey (not the comments) and they do not work. Even the line print("Hallo World!") ist not working, why is this so?

yaleman, 2023-04-25:
How'd you install it, and could you share your code? There's no hello world print statement in the current codebase.

chrmuh, 2023-04-25:
Sorry, I saw the color of the "print" wrong so forget about that.

I attached a picture of your example code and the specific code segment. My problem is that I have no idea why the console isn't printing the battery volume.

And yes there is no hello world in the code, I only added it at line 43 for one single test and it doesn't work eighter.

[image]

yaleman, 2023-04-25:
It looks like you're running `pytest` and expecting `example.py` to run - this is not how `pytest` works. You need to just run `python example.py` 

```shell
➜ python example.py
Single Inverter
Available fields in data: dict_keys(['info', 'kpi', 'powercontrol_status', 'images', 'weather', 'inverter', 'hjgx', 'homKit', 'isTigo', 'tigoIntervalTimeMinute', 'smuggleInfo', 'hasPowerflow', 'powerflow', 'hasGridLoad', 'isParallelInventers', 'isEvCharge', 'evCharge', 'hasEnergeStatisticsCharts', 'energeStatisticsCharts', 'energeStatisticsTotals', 'soc', 'environmental', 'equipment'])
"49.2V/-0.9A/0W"
Battery voltage is: 49.2
```

This assumes you have copied config.py.example to config.py and filled out your details. It used to be mentioned in the README but seems to have been lost - I've updated that now.

closed 2023-06-02 by yaleman