yaleman/pygoodwe #283: GoodWe rate limit question

WouterGritter, 2024-07-08:
I had been using it for a while to fetch data from my solar panels, and after a couple months my IP got banned. The app didn't work on my home network and the inverter was not able to send out data. I had fixed it by tunneling all GoodWe data through a VPN via my router, and after about a year my IP-ban got removed. But I digress.

Have you had any experience with, or is there any documentation on GoodWe's rate limit? I was polling their API 12 times per hour, with 5 minute intervals. I want to use this project again to act as a bridge between my GoodWe solar inverter and a MQTT broker, and I will probably fetch the data once every 30 minutes, and only between certain times when the sun is out as a safety measure.

yaleman, 2024-07-10:
I don't have any information on their rate limits or terms of service sadly. Perhaps you could look at the headers they return to see if they're noting any rate limit info, or when you do get limited, what's returned then?

WouterGritter, 2024-11-19:
I haven't checked yet, but since posting the issue it's been running stable with 15 minute intervals to fetch data from GoodWe's API. I am also only polling the API between sunrise and sunset, so either there's a daily limit or the limit that'll get you IP banned is within 5 and 15 minutes.

closed 2024-11-19 by WouterGritter

timverm-ikdoeict, 2025-10-26:
My IP was banned too. I had contact with their helpdesk and they said the rate limit was 4000 requests in 1 hour. 
After a conversation and me prommising to limit the number of requests, they unbanned my IP.

timverm-ikdoeict, 2025-11-03:
> Lucky you! I could not get in contact with them regarding this.

I contacted the BENELUX after sales support team by mail. After a first row helpdesk employee, I arrived at someone who suggested a phone call. After me promising to limit the number of requests, I got unbanned. 
Contact information (in dutch) can be found at https://nl.goodwe.com/contact-us. I noticed that if I change the language to English, I arrive at a completely different looking site. Maybe this can help you or someone else.